space-ros/stacks #18: Invalid Cobra SARIF output

Ronoman, 2022-07-25:
# Preconditions:
- A built Space ROS docker container
- `colcon test` run in the spaceros workspace

# Expected behavior:
All SARIF output from test generation is valid and can be parsed by standard tooling.

# Actual behavior:
Some Cobra SARIF output is invalid by JSON standards. Take for example `rosidl_typesupport_cpp/test_results/rosidl_typesupport_cpp/cobra-autosar.sarif`, line 92 (line numbers may differ across test runs, unsure):

`"text": "(Required) All code shall conform to ISO/IEC 14882\:2014 -Programming Language C++ and shall not use deprecated features."`

In JSON, `\` is an escape character which can only be followed by specific characters (see https://stackoverflow.com/a/27516892 for a nice diagram), of which `:` is not one. This prevents standard tooling (in my case, Sarif Viewer by Microsoft DevLabs) from properly parsing these files.

mjeronimo, 2022-07-25:
@Ronoman We use the JSON output from the Cobra tool and then use the json_covert tool that comes with Cobra to convert the JSON formatted output to SARIF. So, problems with the SARIF output should be entered at the Cobra github repo: https://github.com/nimble-code/Cobra. Please enter an issue there and we can then refer to that issue here for tracking. 

When entering the issue over there, to make it easy to reproduce, I recommend providing an output JSON file and the command line to json_covert that will then generate the ill-formed SARIF.

Ronoman, 2022-07-27:
Will do -- currently struggling to reproduce this as I'm not getting any SARIF output from Cobra anymore, but will submit an issue there once I am able to.

Ronoman, 2022-08-05:
Added a comment to an issue you already raised: https://github.com/nimble-code/Cobra/issues/49#issuecomment-1206844159.

mjeronimo, 2022-08-24:
@Ronoman Has this been fixed in Cobra? Still an open issue?

mjeronimo, 2022-11-18:
@Ronoman Closing, given your current satisfaction with the Cobra output. Please re-open if you don't agree.

closed 2022-11-18 by mjeronimo