space-ros/stacks #19: Rule texts file not provided to misra tool in cppcheck

Ronoman, 2022-07-27:
In `cppcheck.sarif` files generated by cppcheck through `colcon test`, all `rules.shortDescription.text` are "misra violation (use --rule-texts=<file> to get proper output)". While the actual rule ID that was violated is set in `rules.id`, it would be useful for developers to be able to see the rule violation in the sarif output alone. I suggest 2 changes:

1. Add generation of a rule texts file to the `colcon test` pipeline (alternatively, generate this file apriori and include it with the Space ROS docker image)
2. Provide this file to all runs of misra.py in cppcheck.

Let me know if this path forwards seems reasonable, I can take ownership on a PR for these changes if they make sense.

Ronoman, 2022-07-27:
This is probably the wrong repository for this discussion -- will move issue to https://github.com/ament/ament_lint/tree/spaceros if I get confirmation that is the right place for this discussion.

mjeronimo, 2022-08-24:
@Ronoman I've opened a PR here: https://github.com/ament/ament_lint/pull/403 to allows a user to provide the rule text input to the cppcheck tool. The rule text itself is copyrighted, so this is the best we can do.

closed 2022-09-01 by Ronoman