space-ros/stacks #32: Docker Login resource required?

Pedro-Roque, 2022-10-12:
Terminal output:
```
➜  demo_spaceros git:(main) ./build.sh 

##### Building Space ROS Demo Docker Image #####

Sending build context to Docker daemon  13.82kB
Step 1/31 : FROM openrobotics/spaceros:latest
pull access denied for openrobotics/spaceros, repository does not exist or may require 'docker login': denied: requested access to the resource is denied

##### Done! #####
```
How can one request access to these repos?

Ronoman, 2022-10-12:
The missing `openrobotics/spaceros` docker image can be built by following the README in the spaceros subdirectory in this repository: https://github.com/space-ros/docker/blob/main/spaceros/README.md.

Pedro-Roque, 2022-10-13:
Thanks @Ronoman , fixed by building the spaceros Docker first. Also faced an issue related with GPU availability on docker, which required me to install the nvidia-container-toolkit. Maybe good to add this to the intructions as well, just in case / to a FAQ section

closed 2022-10-13 by Pedro-Roque