space-ros/stacks #34: libEGL warning: DRI2: failed to create dri screen

Pedro-Roque, 2022-10-13:
After running the docker demo, I get the issue mentioned in the title, and no robot or robot screen show up. Probably related with the nvidia drivers?

Host system:
Ubuntu 20.04
RTX3060
NVIDIA-SMI 515.43.04    Driver Version: 515.43.04    CUDA Version: 11.7  

Anyone faced something similar?

Ronoman, 2022-10-13:
You mentioned in #32 that you may have resolved this by installing the `nvidia-container-toolkit` package. What this enough to resolve this issue? Also, was this package installed on your host, or within the Docker container?

Pedro-Roque, 2022-10-13:
> 

It did not, no. Installed on the host. Should I do it in the docker as well?

mkhansenbot, 2023-02-27:
> > 
> 
> It did not, no. Installed on the host. Should I do it in the docker as well?

In my experience with running Nvidia Isaac Sim, the container toolkit is needed for it to access the GPU from the container. I don't know about Ignition / Gazebo. What happens when you run `nvidia-smi`? First run it outside the container, then run it from within. If you get GPU access in both cases, you're probably fine. If not you probably need the toolkit.

quarkytale, 2023-04-19:
Can you run the launch command with debug mode on? 
That libEGL statement is exactly what its printing, a warning and usually can be ignored. See https://github.com/gazebosim/gz-rendering/issues/587

The issue of robot not showing up is due to something else, need more information to debug.

quarkytale, 2023-04-19:
Also I think issue might have been already fixed see https://github.com/space-ros/demos/issues/15