space-ros/stacks #52: Building space_robots demo fails

Victor4X, 2023-06-01:
Building the space_robots docker image currently fails on rosdep install because there is no valid warehouse_ros_mongo package in the Ubuntu Jammy apt repositories for humble anymore.

This can be fixed by manually building the warehouse-ros-mongo package as seen in my commit here: https://github.com/Victor4X/docker/commit/3dcc704d9648ff16fd265266a1eeef0f31f2ca0b

Or as UniversalRobots have done, by replacing warehouse_ros_mongo with warehouse_ros_sqlite: https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/pull/362

mkhansenbot, 2023-07-11:
@Victor4X - do you want to submit a PR with your changes from above? (https://github.com/Victor4X/docker/commit)/3dcc704d9648ff16fd265266a1eeef0f31f2ca0b

If so I will test and review them. If not, I'll submit a PR myself. I'd like to get this resolved either way.

mkhansenbot, 2023-07-11:
Good news, I tried your changes in a branch, and they seem to have fixed the MongoDB issue.

Bad news, I am now hitting a different error. 
```
#0 101.0 --- stderr: ros_gz_bridge
#0 101.0 CMake Error at CMakeLists.txt:81 (find_package):
#0 101.0   By not providing "Findactuator_msgs.cmake" in CMAKE_MODULE_PATH this
#0 101.0   project has asked CMake to find a package configuration file provided by
#0 101.0   "actuator_msgs", but CMake did not find one.
#0 101.0 
#0 101.0   Could not find a package configuration file provided by "actuator_msgs"
#0 101.0   with any of the following names:
#0 101.0 
#0 101.0     actuator_msgsConfig.cmake
#0 101.0     actuator_msgs-config.cmake
#0 101.0 
#0 101.0   Add the installation prefix of "actuator_msgs" to CMAKE_PREFIX_PATH or set
#0 101.0   "actuator_msgs_DIR" to a directory containing one of the above files.  If
#0 101.0   "actuator_msgs" provides a separate development package or SDK, be sure it
#0 101.0   has been installed.
#0 101.0 
#0 101.0 
#0 101.0 ---
#0 101.0 Failed   <<< ros_gz_bridge [7.33s, exited with code 1]
```
@Victor4X  - Can you try to build again and tell me if you get the same error?

alexbuyval, 2023-07-22:
Hi @mkhansenbot 

After using @Victor4X fix I got the same error:
`Could not find a package configuration file provided by "actuator_msgs`

 I found a related [issue](https://github.com/gazebosim/ros_gz/issues/401), so to fix it I have added the following into demo_manual_pkgs.repos:
```
   actuator_msgs:
    type: git
    url: https://github.com/rudislabs/actuator_msgs.git
    version: main
```
 
Now I have successfully build the docker.

mkhansenbot, 2023-07-25:
@alexbuyval - can you submit a PR with the needed changes to fix the build, I will test and review it!

alexbuyval, 2023-07-25:
@mkhansenbot Done

mkhansenbot, 2023-08-10:
@alexbuyval - we have the docker build working now, can you test and confirm? If so, can you close this issue? Thanks!

alexbuyval, 2023-08-12:
@mkhansenbot 
I have just pulled the main branch, then successfully rebuilt the 'moveit' docker. However when I am trying to build 'space_robots' docker I am getting the error again:

```
232.2 ERROR: the following rosdeps failed to install
232.2   apt: command [sudo -H apt-get install -y ros-humble-warehouse-ros-mongo] failed
232.2   apt: Failed to detect successful installation of [ros-humble-warehouse-ros-mongo]
------
Dockerfile:70
--------------------
  69 |     
  70 | >>> RUN sudo apt-get update -y \
  71 | >>> && /bin/bash -c 'source "${SPACEROS_DIR}/install/setup.bash"' \
  72 | >>> && /bin/bash -c 'source "${MOVEIT2_DIR}/install/setup.bash"' \
  73 | >>> && rosdep install --from-paths src --ignore-src -r -y --rosdistro ${ROSDISTRO}
  74 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c sudo apt-get update -y && /bin/bash -c 'source \"${SPACEROS_DIR}/install/setup.bash\"' && /bin/bash -c 'source \"${MOVEIT2_DIR}/install/setup.bash\"' && rosdep install --from-paths src --ignore-src -r -y --rosdistro ${ROSDISTRO}" did not complete successfully: exit code: 1
```
 Maybe I missed something?

mkhansenbot, 2023-08-13:
You're right, now I see that it is also failing in CI but for some reason it is marking the CI as passing. I'll investigate. In the meantime, can you rebase your PR #58 onto main so we can see if it fixes the issue?

mkhansenbot, 2023-08-16:
Fixed by #58

closed 2023-08-16 by mkhansenbot