space-ros/stacks #64: space_robots CI job is failing to build due to an API name change in MTC

EzraBrooks, 2023-07-27:


mkhansenbot, 2023-07-28:
Is this the failure you're referring to? 
https://github.com/space-ros/docker/actions/runs/5682919408/job/15402383975#step:5:9276
```
/home/spaceros-user/moveit2/src/moveit_task_constructor/core/src/solvers/pipeline_planner.cpp:183:22: error: ‘struct planning_interface::MotionPlanResponse’ has no member named ‘trajectory’; did you mean ‘trajectory_’?
```

EzraBrooks, 2023-07-28:
yep, sorry, I should've linked the output

mkhansenbot, 2023-08-08:
Looks like this is fixed in the upstream MTC repo here:
https://github.com/ros-planning/moveit_task_constructor/commit/564804bd7bb9e320b304b8c4880d96c6ce63b915

That commit is in the humble branch. Weird that last week there was no humble branch, it seems they deleted it and re-added it again. In the meantime we reverted the change to using the humble branch because it broke the build (see #59). Now it seems we need to revert the revert!

PR #65 (2023-08-10): Update moveit_task_constructor branch back to *humble*
Attempt to fix #64. The upstream branch was updated back to `humble`.
closed 2023-08-10 by EzraBrooks