space-ros/stacks #66: Refactor docker build of moveit2 to use build.sh

mkhansenbot, 2023-08-10:
Since we're providing a `build.sh` script for users to build the containers, we should use those in the build flow to ensure that the users build script is always working

This is done in #65 for the `space_robots` build, and needs to be done for the `moveit2` build.

EzraBrooks, 2023-12-28:
I've just realized that if we do this, we'll either lose the automatic GitHub Actions caching stuff for the Docker builds or have to create a custom implementation of it.. this might not be a good idea.

mkhansenbot, 2023-12-30:
I'm open to any other way of testing the build.sh scripts. However, I'm not sure what we lose. Did you see the implementation in #65?

EzraBrooks, 2024-01-02:
I didn't realize at the time, but I believe that change in #65 also lost us some caching in CI.

EzraBrooks, 2024-01-11:
We should consider eliminating build.sh in favor of more cleanly encapsulated `Dockerfile`s and `Earthlyfile`s so you just run `docker build` or `earthly build`

mkhansenbot, 2024-01-30:
Added to next milestone (proposed) humble-2024.04.0

ivanperez-keera, 2025-04-11:
Closed as duplicate of https://github.com/space-ros/docker/issues/24.

closed 2025-04-11 by ivanperez-keera