space-ros/stacks #67: Update the space_robots workflow to run the demo simulations

mkhansenbot, 2023-08-10:
I'm not sure if this is possible in GHA, but ideally, we should have a test that runs the demos by using the `run.sh` script to start a headless simulation in gazebo.

ivanperez-keera, 2024-01-10:
I would love to see that happen. I didn't even think about running it headless.

That being said, it looks like this could be non-trivial (and please correct me if I'm wrong). Especially detecting that it's running correctly, since there is no clear end condition.

What do you guys think about making some of the less clear issues (or the ones where it's less clear if we'll have the time/resources to do it) to discussions, to keep the issues list a bit shorter and more manageable?

mkhansenbot, 2024-01-30:
Added to next milestone (proposed) humble-2024.04.0