space-ros/stacks #77: Moveit2 build takes over 2.5 hours in GHA workflow

mkhansenbot, 2023-08-14:
This is a performance issue. The Moveit2 build takes 2.5 hours in the GHA workflow, which makes it very painful to submit PR's and get them integrated in a timely fashion.

The simplest thing to fix this might be to push the image to dockerhub, so that if nothing in the Moveit2 build has changed, it will simply pull the latest image. However, we probably still want to work on getting the build down to an hour or less if possible.

mkhansenbot, 2023-08-14:
Here's a link for more details:
https://github.com/space-ros/docker/actions/runs/5851111270/job/15861418276

mkhansenbot, 2023-08-16:
We can change the build to only build `moveit2` if the `/moveit2` files are changed:
https://docs.github.com/en/actions/using-workflows/events-that-trigger-workflows#pull_request

mkhansenbot, 2023-08-25:
@EzraBrooks - what do you think of the idea of pushing the latest moveit2 build to docker and only re-building it if the /moveit2 files change?

EzraBrooks, 2023-08-28:
I think that makes sense as an immediate measure before we have time/energy to refactor our containers to take good advantage of Docker's layer caching so it handles that for us.

PR #95 (2023-12-28): Don't limit parallel workers in moveit2 build (#77)
Closes #77.

This is a common hack to avoid running out of memory while building a large C++ codebase. However, it's unnecessary in an era where users just need to be educated to configure `zram` on their system.
closed 2023-12-28 by EzraBrooks