space-ros/stacks #79: Don't push the spaceros image every time we submit a PR

mkhansenbot, 2023-08-15:
We currently have one workflow with jobs for Earthly `build-and-test` and Earthly `build-and-push`. This means we have to run both together to run a `build-and-test`. This means we're pushing containers that aren't based on the `main` branch. I think we should split these into separate workflows. The `build-and-test` workflow should run on every PR, every merge to main, and nightly. The build-and-push should only run once or twice a day. In fact, I think `build-and-push` should be `build-test-push`

xfiderek, 2023-12-17:
@mkhansenbot is this issue still relevant? I can't find these jobs. I can only see `build-testing-results` and `space-ros-image`  in [here](https://github.com/space-ros/docker/blob/main/.github/workflows/earthly-build.yaml)

mkhansenbot, 2023-12-30:
Sorry, I just re-looked at this, I named them wrong in the initial issue, you have the names correct. 

The point is still the same, we have one workflow that does both, which means we can't run them separately, which we really should be doing. We don't want to push an image every time there's a PR to this repo. We need to split the action into two.

mkhansenbot, 2024-01-30:
Added to next milestone (proposed) humble-2024.04.0

mkhansenbot, 2024-05-01:
Fixed by #133

closed 2024-05-01 by mkhansenbot