space-ros/stacks #87: Vision_msgs not present when building space_robots image causing it to fail

j-corvo, 2023-11-20:
Hello,

I was trying to build the space_robots image after I successfully built `moveit` and `spaceros` images, but I found the following error in the building process:

[image]

Am I missing any steps prior to this one?

EzraBrooks, 2023-11-28:
Interesting.. I have another project that uses Gazebo that doesn't require explicitly pulling in `vision_msgs`, and this was building fine until very recently. I wonder if there was a breaking change?

KatherineZT, 2023-12-08:
I fixed this issue by adding the following repo in the `demo_manual_pkgs.repos` file of the `space_robots` directory :
```
vision_msgs:
  type: git
  url: https://github.com/ros-perception/vision_msgs.git
  version: ros2
```

EzraBrooks, 2023-12-14:
I just noticed that the scheduled builds on this repo haven't been running due to inactivity - I'll run one to see if we encounter this issue in CI. Could one of you @KatherineZT @j-corvo open a PR with this addition?

Please use a tagged version, not `version: ros2`, thanks!

EzraBrooks, 2023-12-14:
Confirmed that it is broken on main in CI as well: https://github.com/space-ros/docker/actions/runs/7214030223/job/19655225694

PR #97 (2023-12-28): Add vision_msgs to demo deps (#87)
This source dependency is missing and failing the build. Closes #87.
closed 2023-12-28 by EzraBrooks