space-ros/stacks #94: docker/spaceros does not build

yuyuqq, 2023-12-26:
I tryed to build docker/spaceros. 
Following instruction of the getting started section:

```
$ git clone git@github.com:space-ros/docker.git
Cloning into 'docker'...
...
Resolving deltas: 100% (597/597), done.
$ cd docker/spaceros/
$ ./build.sh 

##### Building Space ROS Docker Image #####

 Init 🚀
————————————————————————————————————————————————————————————————————————————————

           satellite | my-satellite is waking up. Please wait...
           satellite | ...avoiding debris...
           satellite | ...calibrating transponders...
           satellite | ...stabilizing trajectory...
           satellite | ...System online.
           satellite | Connecting to my-satellite...
           satellite | ...Done
           satellite | Version github.com/earthly/buildkit v0.7.23 086f60eecf5a2261bbd53299614f88b3def12746
           satellite | Platforms: linux/amd64 (native) linux/amd64/v2 linux/amd64/v3 linux/amd64/v4 linux/386 linux/arm64 linux/riscv64 linux/ppc64 linux/ppc64le linux/s390x linux/mips64le linux/mips64
           satellite | Utilization: 0 other builds, 0/12 op load
           satellite | GC stats: 1.8 GB cache, avg GC duration 0s, all-time GC duration 0s, last GC duration 0s, last cleared 0 B

Streaming logs to https://cloud.earthly.dev/builds/58589fcd-c282-460a-8d6e-c4ae81f05918

...

 Build 🔧
————————————————————————————————————————————————————————————————————————————————

        ubuntu:jammy | --> Load metadata ubuntu:jammy linux/amd64
               +base | --> FROM ubuntu:jammy
               +base | [----------] 100% FROM ubuntu:jammy
              +setup | --> RUN apt-get update
    ...
Error: Earthfile line 97:2 copy artifact: apply build github.com/space-ros/space-ros:earthly-wrapper+repos-file: earthfile2llb for github.com/space-ros/space-ros:earthly-wrapper+repos-file: resolve build context for target github.com/space-ros/space-ros:earthly-wrapper+repos-file: read git-hash: failed to load cache key: repository does not contain ref earthly-wrapper, output: ""
in		github.com/space-ros/docker/spaceros:humble-2023.10.0+sources
View logs at https://cloud.earthly.dev/builds/58589fcd-c282-460a-8d6e-c4ae81f05918
```

It does not start building. 

(FYI. My colleague said this build got through in about two weeks ago.)

xfiderek, 2024-01-08:
Hi @yuyuqq. now the build should work fine. Please try.

ivanperez-keera, 2024-01-10:
If the original problem reported by @yuyuqq is fixed (because the branch is back), then I'd be inclined to close this issue for now. We already have an issue (#90) that will track the general solution in a more principled way.

EzraBrooks, 2024-01-11:
This is temporarily fixed, and a permanent fix is coming soon.

closed 2024-01-11 by EzraBrooks