space-ros/stacks #99: Docker images do not include `ikos`

ivanperez-keera, 2024-01-09:
So, I'm launching the `space_robots` image, which is based on `spaceros:latest`, and they do not include ikos.

They include `ament-ikos`, but not ikos, rendering the former a bit useless.

I think IKOS should be included with Space ROS.

ivanperez-keera, 2024-01-09:
When I tried to install `ikos` inside the `space_robots` image, I also had to install the python3 packages `catkin_pkg` and `empy`.

ivanperez-keera, 2024-01-11:
There are three ways we can go about this:

- Install IKOS from source (a bit annoying but doable and not hard).
- Build a deb package for IKOS. Harder, but cleaner. The effort will lie on the IKOS repo or an accompanying repo. This aligns best with the goals of IKOS.
- Install IKOS with homebrew. This makes installation obvious at the expense of adding a big dependency to the spaceros docker image.

Thoughts?

xfiderek, 2024-01-11:
I would avoid homebrew install. If we will be able to pull exact version that we are interested in from apt then we should go for it, provided that its possible on IKOS side to maintain repo with deb.

ivanperez-keera, 2024-01-11:
I'd be willing to set for compiling from source for this release, in the interest of at least having something in.

I have an action item to create the DEB package.

I could use help with both of these tasks.

ivanperez-keera, 2024-01-15:
By the way, I don't believe we can install packages with snap inside the docker image, but ikos is available as a snap.

ivanperez-keera, 2024-01-15:
We are still working towards providing a DEB, but that will take some time.

But I definitely think it's important that this version of Space ROS include IKOS.

We can for now install it "manually" and replace the deb in the future.

For extra points, we could figure out which dependencies are build only vs run-only, so that we purge those not needed after installation before the docker layer including ikos is "defined".

ivanperez-keera, 2024-01-21:
IKOS has been submitted to Debian `non-free`: https://ftp-master.debian.org/new/ikos_3.2-1.html. It's still not in, but it's moving up the list to be included. We've also requested the change of license at NASA so that it can be included in Debian `main`.

However, we cannot assume that 1) it'll be included before we freeze (Thursday), and 2) the DEB will be immediately installable in Space ROS's docker image (the versions of libraries it is linked against may be different; that's often the case across distros and distro versions versions with, at least, `libc`).

I'd say, for this time around, we just undo the change that Brian pointed to, compile it inside the Space ROS image, and release.

ivanperez-keera, 2024-01-21:
Here I have to disagree.

The first is that it's not last-minute: it's reverting a change that happened before, and we've been talking for a while on how to best do this. I'll also note that the reason why IKOS was removed was due to the LLVM version available in Ubuntu 22.04, not for any reasons related to the compilation time.

The second is that we don't need to run the analysis, only make IKOS available inside the image. Most people will just pull from osrf/space-ros:latest, so it's not something that affects them in any way. If we can compile other packages, we can build IKOS.

ivanperez-keera, 2024-01-23:
I have the DEBs for ikos. I tried it on a fresh ubuntu and they work.

I'm not sure about whether it would be best to make them release artifacts, or to put them elsewhere. But once that's sorted, installing them in the image should be trivial.

ivanperez-keera, 2024-01-25:
If I'm reading the Github Actions right, the build times for the spaceros image are the same with and without IKOS. It doesn't appear to have a bit impact in the build time.

commit cbf81461 (2024-01-26): Merge branch 'dev-ikos' into main. Close #99.
closed 2024-01-26 by ivanperez-keera