space-ros/stacks #102: `space_robots` `Dockerfile` includes TODO note

ivanperez-keera, 2024-01-10:
We should avoid leaving todos in code, that's just a recipe for never doing anything about those issues ever.

The following TODO note should be transformed into an issue, or solved.

https://github.com/space-ros/docker/blob/76cd412a7379dd5a5ec338b7b75659afbbf0c9a1/space_robots/Dockerfile#L49-L60

ivanperez-keera, 2024-01-10:
I can also turn this issue into one that describes the very problem that the TODO is trying to point at. But if it's not trivial to fix, then I'd rather get the TODO removed and then we can deal with the issue separately.

SEC-RKatayama, 2024-01-10:
[like]  片山 諒介 reacted to your message:

https://github.com/space-ros/docker/blob/76cd412a7379dd5a5ec338b7b75659afbbf0c9a1/space_robots/Dockerfile#L49-L60