space-ros/stacks #104: IKOS files are not generated in static code analysis

xfiderek, 2024-01-10:
# Summary
`+build-testing` procedure from [spaceros Earthfile](https://github.com/space-ros/docker/blob/main/spaceros/Earthfile) is invoked in CI produces summary of tests and static code analysis. There are no `ikos.sarif` present. Check the [latest artifact](https://github.com/space-ros/docker/actions/runs/7474000373) saved as part of CI.

To investigate the issue i took a look at logs from `robot_state_publisher` test run. The logs (after slight modifications) are available in `~/spaceros/build/robot_state_publisher/ament_ikos/ikos.txt` file in docker image created by earthly:
```
-- run_test.py: invoking following command in '/home/spaceros-user/spaceros/build/robot_state_publisher':
 - /home/spaceros-user/spaceros/install/ament_ikos/bin/ament_ikos --xunit-file /home/spaceros-user/spaceros/build/robot_state_publisher/test_results/robot_state_publisher/ikos.xunit.xml --sarif-file /home/spaceros-user/spaceros/build/robot_state_publisher/test_results/robot_state_publisher/ikos.sarif /home/spaceros-user/spaceros/build/robot_state_publisher

No marker files found when scanning /home/spaceros-user/spaceros/build/robot_state_publisher

-- run_test.py: return code 0
-- run_test.py: generate result file '/home/spaceros-user/spaceros/build/robot_state_publisher/test_results/robot_state_publisher/ikos.xunit.xml' with failed test
-- run_test.py: verify result file '/home/spaceros-user/spaceros/build/robot_state_publisher/test_results/robot_state_publisher/ikos.xunit.xml'
```

I don't have much experience with ikos but from short investigation it seems that `ament_ikos` is looking for `.ikosbin` files, which to my understanding should be created during package compilation.

Apart from that I noticed that `ikos` is not even installed in the docker image. I think we should fix that by adding `ikos_vendor` package to [ament_ikos](https://github.com/ament/ament_ikos) repo, the same way it is done for cobra in [ament_cobra](https://github.com/ament/ament_cobra/tree/master/ament_cobra) repo

ivanperez-keera, 2024-01-10:
Also related #99 .

ivanperez-keera, 2024-01-11:
Thanks for this investigation. I'm going to continue the discussion on why IKOS is not installed in #99. After we get it in the docker image, we can discuss if the results are being generated.

xfiderek, 2024-01-11:
Btw the section in [README](https://github.com/space-ros/docker/tree/main/spaceros#running-an-ikos-scan) that describes how to run IKOS analysis mentions that executing it on all packages takes "very long time". Maybe we should record this time and decide if we want to run IKOS as part of CI at all / run it on different cadence (e.g. monthly).

ivanperez-keera, 2024-01-11:
I propose a different approach.

IKOS can be configured to tune both the speed and the tests (which affect the potential for false positives).

I agree with measuring how long it takes, but, instead of disabling it, lets figure out a set of settings that is fast enough but provides something useful in terms of assurance.