space-ros/stacks #124: Improve rosdep install

xfiderek, 2024-01-25:
## Summary
We are using rosdep to install non-ROS packages from apt, such as uncrustify, eigen, clang-tidy. I suggest adding two improvements:
1. Add list of installed packages to final artifact produced in CI (ideally with exact versions)
2. Prevent installation of ROS packages via apt (e.g. by failing whenever we are about to install package with `ros-${DISTRO}-` prefix).

mkhansenbot, 2024-01-30:
What about rviz? Do we have to build that from source?

xfiderek, 2024-01-31:
Hi @mkhansenbot, good question. 

For now our core spaceros image does not install RVIZ. I remember we had a brief discussion about having `base` and `desktop` images in the future, but whatever I stated here is applicable to `base`, which imo should exclude stuff like gazebo and rviz (gui tools in general). The proposed improvement will make sure that in core image, everything is traceable and subject to code analysis and tests.

mkhansenbot, 2024-01-31:
I agree we should be able to install from debs for the demos. I just want to make sure whatever change is made to block rosdep doesn't include demo dependencies like rviz.

asimonov, 2024-07-25:
> ## Summary
> 
> We are using rosdep to install non-ROS packages from apt, such as uncrustify, eigen, clang-tidy. I suggest adding two improvements:
> 
>     1. Add list of installed packages to final artifact produced in CI (ideally with exact versions)
> 
>     2. Block installation of ROS packages with rosdep (e.g. by failing whenever we are about to install package with `ros-${DISTRO}-` prefix), as all ROS packages must be compiled from source in spaceros.

Yes, I support both!

for 1 I usually use 'simulated' rosdep output then run it like this (this is example from our main Earthfile):

```
  RUN rosdep install -s -y \
      --from-paths src --ignore-src \
      --rosdistro ${ROSDISTRO} \
      # `urdfdom_headers` is cloned from source, however rosdep can't find it.
      # It is because package.xml manifest is missing. See: https://github.com/ros/urdfdom_headers
      # Additionally, IKOS must be excluded as per: https://github.com/space-ros/docker/issues/99
      --skip-keys "$(tr '\n' ' ' < 'excluded-pkgs.txt') urdfdom_headers ikos" \
      > rosdep-commands.sh && \
      chmod u+x rosdep-commands.sh && \
      ./rosdep-commands.sh
```

asimonov, 2024-07-25:
> Hi @mkhansenbot, good question.
> 
> For now our core spaceros image does not install RVIZ. I remember we had a brief discussion about having `base` and `desktop` images in the future, but whatever I stated here is applicable to `base`, which imo should exclude stuff like gazebo and rviz (gui tools in general). The proposed improvement will make sure that in core image, everything is traceable and subject to code analysis and tests.
> 
> I am not sure though what we should do for demos, where rviz and gazebo may be required. My guess would be that such packages are less critical than core ROS middleware etc, so installing debs may be acceptable, but I am of course happy to discuss it further and hear opinion of other people.

Think we discussed it on technical meeting since that comment was made...

My view is that in the demos we can build rviz and plugins from source (I do it in one of my current dockers, so it is not an issue).

With gazebo unfortunately it should come from apt, building from source seem to be difficult. But carefully to not pull some ros-* debs...

ivanperez-keera, 2025-04-27:
This sounds connected to the changes that @Bckempa just incorporated in move it and the demo images.

Also, I'd like to split this in two tickets, one for each point. Experience shows that, even though each item seems small individually, issues like these tend to stagnate for a long time. There's no harm in splitting it, and it'll allow each issue to move forward more quickly.

asimonov, 2025-04-27:
so, where do we want rviz and gazebo? in `space_robots`?

ivanperez-keera, 2025-04-28:
@asimonov To me, that sounds right. The base `space-ros` image should contain more common packages, and likely be more geared towards the stack that will run the robot software. A second image (be is `space-robots` or something else) could include debugging and simulation tools.

I'm open to hearing dissenting arguments.