space-ros/stacks #129: Add VSLAM to Nav2 demo

mkhansenbot, 2024-01-30:
To make the Nav2 demo more realistic, it should use VSLAM instead of lidar based SLAM and localization

Mechazo11, 2024-08-10:
@mkhansenbot is this milestone still active?

Mechazo11, 2024-08-11:
@Bckempa thanks for the info. I wanted to address this issue by starting to incorporate a [ROS 2 package for ORB SLAM3](https://github.com/Mechazo11/ros2_orb_slam3) but in a recent discussion I had, there appears to be an incompatibility between GPL v3.0 and Apache 2.0 license that Space ROS is based on. Are you familiar with this licensing issue?

ivanperez-keera, 2025-04-27:
This issue seems non-trivial for a number of reasons. I think it's best to move this to a discussion, and open separate issues when we have a specific, achievable plan of action.