space-ros/stacks #131: Upgrade simulator for demos

mkhansenbot, 2024-01-30:
Gazebo Fortress is a bit difficult to use due to the fact it was the version that started out as Ignition and then changed to Gazebo.

We should look at upgrading the demos. While we're at it, it would be good to look at O3DE, Isaac Sim or other high-performance simulators. The current mars_rover demo, for example is very slow.

mkhansenbot, 2024-01-30:
Added to next milestone (proposed) humble-2024.04.0

asimonov, 2024-06-07:
hi Matt,

sorry, i was not able to look at SpaceROS after last meeting in April.
but I am starting to do it now.

and this is my first question - which gazebo to use...

I have just realised this demo is on Fortress.
I think I need at least Garden to work with the Moon example model that I am interested in.
But looks like Gazebo Harmonic is out now and it is LTS. 
So I will try examples I am working on with Harmonic...

Would you consider moving this Mars demo to Harmonic?

asimonov, 2024-06-12:
@Bckempa 
I have done some naive upgrade of moveit2 and space_robots to Gazebo Harmonic. 

for some reason rosdep does not install all dependencies for Harmonic automatically, probably because it is not official Humble simulator version. But I end up with both Fortress and Harmonic simulators in the docker in the end. 

Mars demo does not run on Harmonic out of the box, so now trying to debug why.

asimonov, 2024-06-13:
My upgrade to Gazebo Harmonic works with mars_rover demo. I did not check canadarm demo.

The change is spread over 3 repos: docker, demos and simulation.

docker: 
https://github.com/space-ros/docker/commit/5103c8999bfb30cdb5bee14e1d5565c59fa5ec44

simulation:
https://github.com/space-ros/simulation/commit/622cdc6eb36dd22ac9540f6d836c4af7803a8a78

demos:
https://github.com/asimonov/demos/commit/68c9195e42eed166c3d3138108f479819bc21389
https://github.com/asimonov/demos/commit/3a00ea1eab7295cb18920d45d899c33516405487

For some reason setting up env variable inside launch python does not work, so have to use this before launching:

export GZ_SIM_RESOURCE_PATH=/home/spaceros-user/demos_ws/install/ros_gz_sim_demos/share:/home/spaceros-user/demos_ws/install/simulation/share/simulation/models:/home/spaceros-user/demos_ws/install/simulation/share

If there is interest in merge, we can discuss how to do it. I am not sure if the plan is to refactor docker images first or not...