space-ros/stacks #138: Running an IKOS scan from tutorial does not work

xfiderek, 2024-02-20:
After trying to run IKOS scan as per [this tutorial](https://github.com/space-ros/docker/tree/main/spaceros#running-an-ikos-scan), I've encountered 3 problems.

### Problem A: colcon build command throws basic errors 

Following the readme, I try to execute

```bash
spaceros-user@d10d85c68f0e:~/spaceros$ CC="ikos-scan-cc" CXX="ikos-scan-c++" LD="ikos-scan-cc" colcon build --build-base build_ikos --install-base install_ikos --cmake-args -DSECURITY=ON -DINSTALL_EXAMPLES=OFF -DCMAKE_EXPORT_COMPILE_COMMANDS=ON --no-warn-unused-cli
```

First impediment is that `ikos-scan-cc` executable is missing, which can easily be fixed. Apart from that, we also need to set IKOS_SCAN_NOTIFIER_FILES env variable in the container. Otherwise it throws: 

```bash
error: ikos-scan-cc: missing environment variable IKOS_SCAN_SERVER or IKOS_SCAN_NOTIFIER_FILES
```

### Problem B: building `cobra_vendor` package fails when `ikos-scan-cc` is used as a compiler

The exact error thrown by `cobra_vendor` package is:

```bash
# compile command
ikos-scan-cc -Wall -pedantic -Werror -Wshadow -O2 -DYY_NO_INPUT -I. -std=c99 -o scope_check scope_check.c cwe_util.c c.ar -pthread
# error
clang: error: c.ar: 'linker' input unused [-Werror,-Wunused-command-line-argument]
```

c.ar is a library that is build earlier and which should be linked at this step.

Changing the end of compile command from `c.ar -pthread` to `-Wl,c.ar -pthread` fixes the issue, so i presume there is something wrong with how `ikos-scan-cc` forwards arguments to clang. The easiest will be to fix that in ament’s `cobra_vendor` package. Two other alternatives are to either fix it in `ikos` side or directly in `cobra`.

 

### Problem C: `ament _ikos` timeouts for large packages (e.g. rclcpp)

The reason is that `ament_ikos` uses default timeout time of 60 seconds. We should modify this value in the repo. 

After addressing these 3 issues, I was able to run ikos scan as expected in spaceros workspace. I will create 3 PRs, first for this repo (problem A), second for `ament_cobra` repo (problem B), and third for `ament_ikos` repo (problem C)

xfiderek, 2024-02-20:
@ivanperez-keera, tagging you for visibility.

I have created 3 PRs:
Problem A: https://github.com/space-ros/docker/pull/139
Problem B: https://github.com/ament/ament_cobra/pull/16
Problem C: https://github.com/ament/ament_ikos/pull/4

ivanperez-keera, 2024-02-20:
This is extremely useful! Thank you @xfiderek for doing this.

ivanperez-keera, 2024-02-20:
(I'll review in more detail later but I wanted to send you that quick note :smiley: )

commit 3bbbaa20 (2024-04-26): Merge branch 'xfiderek/ikos'. Close #138.
closed 2024-04-26 by ivanperez-keera