space-ros/stacks #142: Need to pin the Navigation2 version number for stability

mkhansenbot, 2024-04-14:
Issue #140 is now fixed but in order to prevent future breakage, we should pin the version number for Navigation2 to a known good version, the same way we do with other repos.

xfiderek, 2024-04-14:
Hi @mkhansenbot , I've noticed the same and have just kicked off the discussion to make it robust: https://github.com/space-ros/space-ros/discussions/148

xfiderek, 2024-05-28:
See discussion https://github.com/space-ros/space-ros/discussions/140 and issue https://github.com/space-ros/space-ros/issues/107 for more context
TODOs
* Use a new generate_repos.sh file from updated spaceros dev image and generate pinned .repos file
* Move the docker image to `stacks` repo
* Decide if we want to push this image to dockerhub

asimonov, 2024-08-08:
May I ask what is the version of nav2 we would like to pin?

I am currently trying to create a version of this nav2 docker for my use. I thought jazzy version of nav2 (released few weeks back) would be a great idea to try... Looks like it is not. 

what I tried:
* take our current humble space-ros base image
* clone nav2/jazzy
* play around with nav2 dependencies - try jazzy, but switch to humble if it does not work
* make sure no ros-* apt packages get into the build, so manually remove everything related to gazebo, rviz, slam_toolbox and few others
* getting to nav2 colcon build

The nav2 build is failing due to core ros2 package versions incompatibilities. For example:

```
--- stderr: nav2_util                               
In file included from /home/spaceros-user/nav2_ws/src/navigation2/nav2_util/src/base_footprint_publisher.cpp:17:
/home/spaceros-user/nav2_ws/src/navigation2/nav2_util/src/base_footprint_publisher.hpp:57:8: error: ‘void nav2_util::BaseFootprintPublisherListener::subscription_callback(tf2_msgs::msg::TFMessage_<std::allocator<void> >::ConstSharedPtr, bool)’ marked ‘override’, but does not override
   57 |   void subscription_callback(tf2_msgs::msg::TFMessage::ConstSharedPtr msg, bool is_static) override
      |        ^~~~~~~~~~~~~~~~~~~~~
/home/spaceros-user/nav2_ws/src/navigation2/nav2_util/src/base_footprint_publisher.hpp: In member function ‘void nav2_util::BaseFootprintPublisherListener::subscription_callback(tf2_msgs::msg::TFMessage_<std::allocator<void> >::ConstSharedPtr, bool)’:
/home/spaceros-user/nav2_ws/src/navigation2/nav2_util/src/base_footprint_publisher.hpp:59:45: error: ‘void tf2_ros::TransformListener::subscription_callback(tf2_msgs::msg::TFMessage_<std::allocator<void> >::ConstSharedPtr, bool)’ is private within this context
   59 |     TransformListener::subscription_callback(msg, is_static);
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
In file included from /home/spaceros-user/nav2_ws/src/navigation2/nav2_util/src/base_footprint_publisher.hpp:25,
                 from /home/spaceros-user/nav2_ws/src/navigation2/nav2_util/src/base_footprint_publisher.cpp:17:
/opt/spaceros/install/tf2_ros/include/tf2_ros/tf2_ros/transform_listener.h:158:8: note: declared private here
  158 |   void subscription_callback(tf2_msgs::msg::TFMessage::ConstSharedPtr msg, bool is_static);
      |        ^~~~~~~~~~~~~~~~~~~~~
gmake[2]: *** [src/CMakeFiles/base_footprint_publisher.dir/build.make:76: src/CMakeFiles/base_footprint_publisher.dir/base_footprint_publisher.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:208: src/CMakeFiles/base_footprint_publisher.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
```

i thought i could hack around, but with issues like these I do not think it is possible to jump to jazzy nav2 without moving space-ros to jazzy. FYI nav2 humble is ~650 commits behind jazzy. 

we can pin nav2 repo to humble for now?

xfiderek, 2024-10-31:
Yes, its done

asimonov, 2024-10-31:
> This seems to have been addressed by #152, are we good to close @xfiderek, @eholum?

Yes_____________________________Kind Regards,Alexey Simonov

closed 2024-10-31 by ivanperez-keera