space-ros/stacks #159: Low RTF using gazebo without hardware rendering [NASA Space ROS Sim Summer Sprint Challenge]

dirksavage88, 2024-08-09:
There is a very low (approximately 0.08-0.11 RTF) as shown on the curiosity demo in the default world file.

[image]

Additionally when limited to CPU, the render shows a lot of aliasing possibly due to the lower resolution/CPU rendering constraints.

CPU is an intel i7 1165g7

This is without any form of rendering hardware acceleration. 

**Possible solution: bring in correct correct drivers to enable hardware acceleration and increase the RTF to enable a smoother operating simulation environment (for primarily AMD and Intel, nvidia could be covered under a different issue)**

Mechazo11, 2024-08-10:
@dirksavage88 I would like to reach out to you over email. Would you kindly share it here?

dirksavage88, 2024-08-10:
> @dirksavage88 I would like to reach out to you over email. Would you kindly share it here?
dirksavage88@gmail.com