space-ros/stacks #160: moveit2 image build error: exit code 128, Remote branch moveit2/src not found in upstream origin

Mechazo11, 2024-08-11:
Hello,

Instructions to build the ```osrf/moveit2``` image as part of the build instructions for ```space_robots``` image fails after following the instructions from the latest version of ```docker``` repo

### System
* Ubuntu 22.04, ROS 2 Humble 
* Docker installed and given super user privileges 

#### Steps to reproduce

* Ensured ```osrf/space-ros``` image is available ```docker image list```

[image]

* Build ```moveit2``` image
```
cd ~
mkdir space_ros && cd space_ros
git clone https://github.com/space-ros/docker.git
cd moveit2
./build.sh
```

#### Expected behavior 

```moveit2``` image builds and is available

#### Actual behavior

Build fails. Console output shown below

```

[+] Building 0.6s (13/30)                                                                              docker:default
 => [internal] load build definition from Dockerfile                                                             0.0s
 => => transferring dockerfile: 6.14kB                                                                           0.0s
 => [internal] load metadata for docker.io/osrf/space-ros:latest                                                 0.0s
 => [internal] load .dockerignore                                                                                0.0s
 => => transferring context: 2B                                                                                  0.0s
 => [stage-0  1/26] FROM docker.io/osrf/space-ros:latest                                                         0.0s
 => [internal] load build context                                                                                0.0s
 => => transferring context: 186B                                                                                0.0s
 => CACHED [stage-0  2/26] RUN mkdir /opt/spaceros/src   && vcs import /opt/spaceros/src < /opt/spaceros/exact.  0.0s
 => CACHED [stage-0  3/26] RUN --mount=type=cache,target=/var/cache/apt,sharing=locked   --mount=type=cache,tar  0.0s
 => CACHED [stage-0  4/26] RUN --mount=type=cache,target=/var/cache/apt,sharing=locked   --mount=type=cache,tar  0.0s
 => CACHED [stage-0  5/26] RUN rosdep update                                                                     0.0s
 => CACHED [stage-0  6/26] RUN --mount=type=cache,target=/var/cache/apt,sharing=locked   --mount=type=cache,tar  0.0s
 => CACHED [stage-0  7/26] RUN python3 -m pip install -U   argcomplete   flake8-blind-except   flake8-builtins   0.0s
 => CACHED [stage-0  8/26] WORKDIR /home/spaceros-user                                                           0.0s
 => ERROR [stage-0  9/26] RUN sudo git clone https://github.com/ros-planning/moveit2.git -b ${ROSDISTRO} moveit  0.6s
------                                                                                                                
 > [stage-0  9/26] RUN sudo git clone https://github.com/ros-planning/moveit2.git -b ${ROSDISTRO} moveit2/src:        
0.235 Cloning into 'moveit2'...
0.598 fatal: Remote branch moveit2/src not found in upstream origin
------
Dockerfile:94
--------------------
  92 |     # Get the MoveIt2 source code
  93 |     WORKDIR ${HOME_DIR}
  94 | >>> RUN sudo git clone https://github.com/ros-planning/moveit2.git -b ${ROSDISTRO} moveit2/src
  95 |     RUN cd ${MOVEIT2_DIR}/src \
  96 |       && sudo git clone https://github.com/ros-planning/moveit2_tutorials.git -b ${ROSDISTRO}
--------------------
ERROR: failed to solve: process "/bin/sh -c sudo git clone https://github.com/ros-planning/moveit2.git -b ${ROSDISTRO} moveit2/src" did not complete successfully: exit code: 128

```

Mechazo11, 2024-08-11:
It appears ```nav2``` and ```moveit``` were [moved away from](https://discourse.ros.org/t/move-of-nav2-and-moveit-repositories-at-github/37450) ```ros-planning``` repository a few months earlier.

eholum, 2024-08-11:
Yes, this was changed in https://github.com/space-ros/space-ros/pull/170/files#diff-2705f6016a64189abd14041e2f3e8dbb829553295d9088e4eb42eccb5e61bf47R18 and I foolishly didn't have the required changes queued up here before that got merged. I'm working through testing fixes for this in: https://github.com/space-ros/docker/pull/162/files.

eholum, 2024-08-11:
> It appears nav2 and moveit were [moved away from](https://discourse.ros.org/t/move-of-nav2-and-moveit-repositories-at-github/37450) ros-planning repository a few months earlier.

I think this is unrelated to the issue that you're running into now. Those redirects should be in place but we need to update those references, too.

eholum, 2024-08-11:
This is an issue if you build the base space ros image locally from main, and then try to build this from the latest main. We should rethink tagging locally built images with the same name/tag as what gets pushed to the container registry! Apologies for the trouble as we are currently working through some significant changes to the build order here.

@Mechazo11 I think to fix your issue you have two choices:

1) If you want to stay on the latest main/cutting edge you can try checking out the branch from https://github.com/space-ros/docker/pull/162 and building locally.
2) Remove or move your locally built spaceros image so that it no longer has tag `osrf/space-ros:latest` -
```
# Optionally rename the locally built tag
$ docker image tag osrf/space-ros:latest osrf/space-ros:latest-main

# Remove the existing locally built image
$ docker image rm osrf/space-ros:latest

# Rebuild the moveit2 image using the latest tagged (and correct) version
$ cd <DOCKER_REPO>/moveit2
$ ./build.sh
```

Let me know if you give one or both of those a shot... and what other issues might pop up for you.

Mechazo11, 2024-08-11:
@eholum thank you very much for the possible solutions. I will try the 2nd route and report back if the issue has been resolved or not.

@Bckempa thank you very much for taking the time in looking in to multiple issues I have opened today.

Mechazo11, 2024-08-12:
@eholum Solution 2 worked. ```moveit2``` image took nearly 31 minutes and ```space_robot``` images took 16 minutes to build. I was then able to successfully launch the curiosity rover demo. Steps I took are as follows

```

cd ~
mkdir space_ros && cd space_ros
git clone https://github.com/space-ros/space-ros.git
cd space_ros
./build.sh # Build osrf/space-ros image
./run.sh # Test osrf/space-ros image
exit # Exit out of the docker container
docker image tag osrf/space-ros:latest osrf/space-ros:latest-main # retag image name
docker image rm osrf/space-ros:latest # Remove old tag
cd ..
git clone https://github.com/space-ros/docker.git
cd docker
cd moveit2
sudo ./build.sh
cd ..
cd space_robots
./build.sh
xhost +local:docker # Allow GUI pass through
./run.sh
ros2 launch mars_rover mars_rover.launch.py
``` 

I have a few closing remarks / questions.

1. Once ```space_robot``` image is built, ```moveit2``` image may no longer be necessary. it takes up about 12 Gb of space. I deleted it and was still able to run ```space_robot``` image without any issue.

2. Is the entirety of ```moveit2``` necessary?

Thanks again to the Space ROS team members for helping me in solving this issue. Have a wonderful weekend.

With best,
@Mechazo11

closed 2024-08-12 by Mechazo11

eholum, 2024-08-19:
> Once space_robot image is built, moveit2 image may no longer be necessary. it takes up about 12 Gb of space. I deleted it and was still able to run space_robot image without any issue.

Unfortunately, because the `space_robot` image builds on the [moveit2](https://github.com/space-ros/docker/blob/main/space_robots/Dockerfile#L22) image, removing that image will have little impact on the amount of storage consumed. The date is mostly still there, but the tag will not be. Fortunately, Docker is smart about not duplicated data, so at least having both the `moveit2` and `space_robot` images doesn't actually take up ~24GB (or whatever it is...).

> Is the entirety of moveit2 necessary?

For now, yes. In the near future I'm hoping to be able to push the images for both MoveIt2 and Nav2, so that people no longer have to deal with those long builds...