space-ros/stacks #161: Adjust READMEs and Demos Updated Base Image

eholum, 2024-08-11:
We need to update a handful of references in the README and demo build files to handle folder moves and other changes from https://github.com/space-ros/space-ros/issues/177, including those pushed in https://github.com/space-ros/space-ros/pull/170.

There are still a few hard coded `~/spaceros` lying around, and we standardized some env variables (`ROS_DISTRO` instead of `ROSDISTRO`).

martincerven, 2024-08-19:
Bump, wanted to try docker images as part of **NASA Space ROS Sim Summer Sprint Challenge**, only to find out you refactored a LOT just weeks ago and repo is currently in broken state, but looks like https://github.com/space-ros/docker/pull/162 fixes these ENV mismatches.

eholum, 2024-08-19:
@martincerven Yep, sorry about that. The timing of taking on some significant reworking of our packaging + the challenge could definitely have been better.

If you want to move forward with either the latest release or the fix on main you can follow the instructions in https://github.com/space-ros/docker/issues/160#issuecomment-2282840466.

PR #189 (2024-10-01): Fix remaining issues in applications and demos from the updated spaceros base image
Resolves #161.

There are still issues building `main` for moveit and nav2 after migration of the base image build to the space-ros repo. This PR fixes them, including

* Updating changed environment variable names
* Fixes references to moved or removed directories in the base image
* Parameterizes the base image name for building locally and in CI

As the base `main` image is now pushed to GHCR, we should use that when building in CI. The change can be tested locally if you have access to the container registr [... cut]
closed 2024-10-01 by eholum