space-ros/stacks #165: refactor navigation2 docker - cleanup dependencies

asimonov, 2024-08-12:
I suggest we cleanup navigation2 docker build, still keeping things on humble.
But remove gazebo and rviz dependencies from it.

Gazebo is better to install separately at a later stage, with a view of moving to Gazebo Harmonic. Currently gazebo classic is installed as dependency of nav2_system_tests.

Rviz is currently installed from apt and brings with it a lot of ros-humble*.

xfiderek, 2024-08-13:
hi @asimonov . this will be addressed as part of this https://github.com/space-ros/docker/issues/142

asimonov, 2024-08-15:
@xfiderek yes, I discovered #142 after doing this work. it was created a while ago and from the name I thought it was just about selecting more stable release versions of repos, not refactor of entire dependency install.

I kind of like how you simplified dependencies of nav2 in PR #152 

may be i will close/retract this one. let me do some testing of #152...

asimonov, 2026-10-18:
lets close this as I think most of the work has happened on other PRs

closed 2026-10-18 by asimonov