space-ros/stacks #172: How to run demos on a remote display?

gmacario, 2024-09-03:
I would like to be able to run the [Space ROS Space Robots Demo Docker](https://github.com/space-ros/docker/tree/main/space_robots) on a powerful headless server (hw2482) while displaying the Gazebo GUI on my local machine (delta).

Both hosts are running Ubuntu-desktop 24.04 and can communicate via SSH.

Unfortunately I have been unable to accomplish my goal so far.
Here is what I tried

On my local workstation (HW2228) I first log into the remote server using `ssh -Y hw2482`

If I run some X client application on the server (for instance, `xlogo` or `xeyes`), they are correctly displayed on my local machine:

[image]

However, if I execute the following instructions (from https://github.com/space-ros/docker/tree/main/space_robots#running-the-demos) to run the container on the remote server

```bash
# Logged in as gmacario@delta
ssh -Y hw2482

# Logged in as gmacario@hw2482
cd ~/github/space-ros/docker/space_robots
xhost +local:docker
./run.sh

# Inside container on hw2482
ros2 launch mars_rover mars_rover.launch.py
```

I got the `X11 connection rejected because of wrong authentication.`  error as shown below:

```text
gmacario@delta:~$ ssh -Y hw2482
Welcome to Ubuntu 24.04.1 LTS (GNU/Linux 6.8.0-41-generic x86_64)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/pro

Expanded Security Maintenance for Applications is not enabled.

0 updates can be applied immediately.

8 additional security updates can be applied with ESM Apps.
Learn more about enabling ESM Apps service at https://ubuntu.com/esm

Last login: Tue Sep  3 22:41:57 2024 from 100.69.53.95
gmacario@hw2482:~$ cd ~/github/space-ros/docker/space_robots
gmacario@hw2482:~/github/space-ros/docker/space_robots$ xhost +local:docker
non-network local connections being added to access control list
xhost:  must be on local machine to add or remove hosts.
gmacario@hw2482:~/github/space-ros/docker/space_robots$ ./run.sh
spaceros-user@hw2482:~/demos_ws$ ros2 launch mars_rover mars_rover.launch.py
[INFO] [launch]: All log files can be found below /home/spaceros-user/.ros/log/2024-09-03-20-51-03-883237-hw2482-91
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ign gazebo-1]: process started with pid [92]
[INFO] [robot_state_publisher-2]: process started with pid [94]
[INFO] [create-3]: process started with pid [96]
[INFO] [move_arm-4]: process started with pid [98]
[INFO] [move_mast-5]: process started with pid [100]
[INFO] [move_wheel-6]: process started with pid [102]
[INFO] [run_demo-7]: process started with pid [104]
[INFO] [odom_tf_publisher-8]: process started with pid [106]
[INFO] [parameter_bridge-9]: process started with pid [108]
[INFO] [image_bridge-10]: process started with pid [110]
[create-3] [ros_ign_gazebo] is deprecated! Redirecting to use [ros_gz_sim] instead!
[create-3] 
[robot_state_publisher-2] Warning: link 'lidar_link' material 'black' undefined.
[robot_state_publisher-2]          at line 84 in /home/spaceros-user/spaceros/src/urdfdom/urdf_parser/src/model.cpp
[robot_state_publisher-2] Warning: link 'lidar_link' material 'black' undefined.
[robot_state_publisher-2]          at line 84 in /home/spaceros-user/spaceros/src/urdfdom/urdf_parser/src/model.cpp
[... 86 lines cut ...]
[ign gazebo-1] #10   Object "/usr/lib/x86_64-linux-gnu/libignition-gazebo6-gui.so.6", at 0x745f7443d3a0, in ignition::gazebo::v6::gui::createGui(int&, char**, char const*, char const*, bool, char const*, int, char const*)
[ign gazebo-1] #9    Object "/usr/lib/x86_64-linux-gnu/libignition-gui6.so.6", at 0x745f74e39068, in ignition::gui::Application::Application(int&, char**, ignition::gui::WindowType)
[ign gazebo-1] #8    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x745f72cdecec, in QApplicationPrivate::init()
[ign gazebo-1] #7    Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x745f71bc6b6f, in QGuiApplicationPrivate::init()
[ign gazebo-1] #6    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x745f734eeb16, in QCoreApplicationPrivate::init()
[ign gazebo-1] #5    Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x745f71bc3c07, in QGuiApplicationPrivate::createEventDispatcher()
[ign gazebo-1] #4    Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x745f71bc3712, in QGuiApplicationPrivate::createPlatformIntegration()
[ign gazebo-1] #3    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x745f732beba2, in QMessageLogger::fatal(char const*, ...) const
[ign gazebo-1] #2    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x745f78ec87f2, in abort
[ign gazebo-1] #1    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x745f78ee2475, in raise
[ign gazebo-1] #0    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x745f78f369fc, in pthread_kill
[ign gazebo-1] Aborted (Signal sent by tkill() 198 1000)
[create-3] [INFO] [1725396665.205495759] [ros_gz_sim]: Waiting messages on topic [robot_description].
[create-3] [INFO] [1725396665.210348558] [ros_gz_sim]: Requested creation of entity.
[create-3] [INFO] [1725396665.210410008] [ros_gz_sim]: OK creation of entity.
[odom_tf_publisher-8] [INFO] [1725396665.250909230] [odom_tf_publisher]: Starting odometry_tf_publisher node
[INFO] [create-3]: process has finished cleanly [pid 96]
[INFO] [ros2-11]: process started with pid [349]
[INFO] [ign gazebo-1]: process has finished cleanly [pid 92]
```

The error makes me believe that the `xhost +local:docker` command was not sufficient.

Anybody has ever tried such use case?
What I am doing wrong?

EzraBrooks, 2024-09-04:
It's possible that `xhost +local:docker` doesn't work in this case because it's _not_ local since you're using SSH, but I'm not sure.

What happens if you just run `xhost +`, thereby disabling the ACL entirely? It's not good practice for security reasons to do that on a regular basis, but it will help us diagnose where the issue is with our/your configuration.

EzraBrooks, 2024-09-04:
I've opened issue #173 describing one of the possible causes of this issue. I'm moving this to a discussion since it's more of a question than an issue. I'll keep responding here as we work together to debug this. Thanks!