space-ros/stacks #176: moveit2 image failing to build

ana-GT, 2024-09-05:
### Operating System

Ubuntu 20.04

### Installation type

Local source build

### Version or commit hash

main

### RMW Implementation

FastRTPS

### Client library (if applicable)

Other

### Steps to reproduce issue

Tried to build the **moveit2** image (in order to build the **space-robots** image). Followed the standard instructions here: https://github.com/space-ros/docker/tree/main/moveit2 (in other words, just run  the **./build.sh** script). I am using the space-ros docker image pulled from Github as base image.

The script fails while trying to build the **geometric_shapes** package. Terminal output below. I have built the moveit2 / space-robots images in the past without an issue.

### Expected behavior

After running the ./build.sh script, the moveit2 docker image should be built.

### Actual behavior

The script ./build.sh fails midway. It cannot compile geometric-shapes, according to the terminal output.

### Backtrace or Console output

63.66 Starting >>> geometric_shapes
69.97 Finished <<< rviz_marker_tools [14.4s]

<<< snip, snip, snip output that is not relevant to the error

147.0 Finished <<< filters [1min 44s]
147.0 Starting >>> control_toolbox

<<< snip, snip, snip output that is not relevant to the error

189.5 --- stderr: geometric_shapes
189.5 In file included from /usr/include/eigen3/Eigen/Core:294,
189.5                  from /usr/include/eigen3/Eigen/Geometry:11,
189.5                  from /home/spaceros-user/moveit2/src/geometric_shapes/include/geometric_shapes/aabb.h:34,
189.5                  from /home/spaceros-user/moveit2/src/geometric_shapes/include/geometric_shapes/bodies.h:35,
189.5                  from /home/spaceros-user/moveit2/src/geometric_shapes/src/bodies.cpp:31:
189.5 In member function ‘Eigen::PlainObjectBase<Derived>& Eigen::PlainObjectBase<Derived>::operator=(Eigen::PlainObjectBase<Derived>&&) [with Derived = Eigen::Matrix<double, 3, 1>]’,
189.5     inlined from ‘Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>& Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>::operator=(Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>&&) [with _Scalar = double; int _Rows = 3; int _Cols = 1; int _Options = 0; int _MaxRows = 3; int _MaxCols = 1]’ at /usr/include/eigen3/Eigen/src/Core/Matrix.h:281:22,
189.5     inlined from ‘virtual void bodies::ConvexMesh::useDimensions(const shapes::Shape*)’ at /home/spaceros-user/moveit2/src/geometric_shapes/src/bodies.cpp:852:46:
189.5 /usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:504:7: warning: ‘*(Eigen::DenseStorage<double, 3, 3, 1, 0>*)((char*)&<unnamed> + offsetof(Eigen::Vector3d, Eigen::Matrix<double, 3, 1, 0, 3, 1>::<unnamed>.Eigen::PlainObjectBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> >::<unnamed>.Eigen::MatrixBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> >::<unnamed>.Eigen::DenseBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> >::<unnamed>.Eigen::DenseCoeffsBase<Eigen::Matrix<double, 3, 1, 0, 3, 1>, 3>::<unnamed>.Eigen::DenseCoeffsBase<Eigen::Matrix<double, 3, 1, 0, 3, 1>, 1>::<unnamed>.Eigen::DenseCoeffsBase<Eigen::Matrix<double, 3, 1, 0, 3, 1>, 0>::<unnamed>))’ may be used uninitialized [-Wmaybe-uninitialized]
189.5   504 |       m_storage = std::move(other.m_storage);
189.5       |       ^~~~~~~~~
189.5 /home/spaceros-user/moveit2/src/geometric_shapes/src/bodies.cpp: In member function ‘virtual void bodies::ConvexMesh::useDimensions(const shapes::Shape*)’:
189.5 /home/spaceros-user/moveit2/src/geometric_shapes/src/bodies.cpp:852:46: note: ‘<anonymous>’ declared here
189.5   852 |   mesh_data_->mesh_center_ = Eigen::Vector3d();
189.5       |                                              ^
189.5 c++: fatal error: Killed signal terminated program cc1plus
189.5 compilation terminated.
189.5 gmake[2]: *** [test/CMakeFiles/test_ray_intersection.dir/build.make:76: test/CMakeFiles/test_ray_intersection.dir/test_ray_intersection.cpp.o] Error 1
189.5 gmake[1]: *** [CMakeFiles/Makefile2:441: test/CMakeFiles/test_ray_intersection.dir/all] Error 2
189.5 gmake: *** [Makefile:146: all] Error 2
189.5 ---
189.5 Failed   <<< geometric_shapes [2min 6s, exited with code 2]

MinahilRaza, 2024-09-06:
I had a similar issue a while ago. Can you check your ram usage during the build using `free -mh` in your terminal
This error typically happens when the RAM is full. A work around is to add swap memory.

ivanperez-keera, 2025-04-08:
Gentle nudge.

@ana-GT  Is this still an issue?

ivanperez-keera, 2025-04-27:
@ana-GT  following up on this. The moveit2 image has changed, and it's currently building to completion in CI and on our own machines.

mkhansenbot, 2025-07-29:
Closing this issue as stale

closed 2025-07-29 by mkhansenbot