space-ros/stacks #191: Restructure the contents of the docker repository

franklinselva, 2024-09-26:
### Feature description

We are moving few contents of the `docker` repository to the demos repository as a part of space-ros/space-ros#178.

This will require the current structure to be updated to the new format and I am proposing new directory structure.

The soon to be `stacks` repository can be of the format as described below.

```bash
├── build.sh
├── Dockerfile.moveit2
├── Dockerfile.nav2
├── moveit2
│   ├── entrypoint.sh
│   ├── excluded-pkgs.txt
│   ├── moveit2-pkgs.txt
│   ├── moveit2_tutorials.repos
│   ├── octomap_fix.diff
│   ├── README.md
│   └── resources
├── navigation2
│   ├── entrypoint.sh
│   ├── navigation2.repos
│   └── README.md
├── README.md
├── renode_rcc
│   ├── build.sh
│   ├── config
│   ├── Dockerfile
│   ├── README.md
│   ├── renode-rtems-leon3
│   └── run.sh
├── rtems
│   ├── build.sh
│   ├── Dockerfile
│   ├── example
│   ├── hello
│   ├── hello_posix
│   ├── README.md
│   ├── run.sh
│   └── tinyxml2
├── run.sh
└── zynq_rtems
    ├── build_dependencies.sh
    ├── compile_demos.sh
    ├── Dockerfile
    ├── enter_container.sh
    ├── hello_network
    ├── hello_zenoh
    ├── README.md
    ├── start_network_tap.sh
    ├── stop_network_tap.sh
    └── toolchain.cmake
```
This could be a breaking change for some of the other issues but it should help will growing stacks for the project.

### Implementation considerations

The key changes in the above proposal:
 - The `space_robots` directory related to the demos have been removed.
 - The `build.sh` and `run.sh` has been moved to root of the repository. This will allow stack images which doesn't need special process like the GUI variant as mentioned in #188 
 - The build and run context of the stack images could be something like this,
```bash
# To build nav2 image for example,
docker build -t osrf/space-ros:nav2 -f Dockerfile.nav2 navigation2
```

franklinselva, 2024-09-26:
I would like to see the purpose of other directories like `rtems`, `zynq_rterms` and `renode_cc`. Can someone brief the necessity of these stacks?

franklinselva, 2024-09-26:
Got you. So these are essentially demos on how to build on different hardware and software standpoints?

franklinselva, 2025-06-12:
For the `rtems` demos. I want to see if we can create a new repository targetting hardware related demo. I think it would be beneficial to decouple simulation and hardware since not all will use the demos of both simulation and hardware. If required, we can pull two separate repositories.

But its more of my thought and wanted to see if this is something to agree.

Bckempa, 2025-11-13:
Ok, now we need to decide where to put the other demos so we can rename this stacks. @franklinselva suggested a separate "hardware demos" repo, we can explore other options as well. Potential topic for discussion at the [technical meeting](https://github.com/space-ros/space-ros/discussions/318)