space-ros/stacks #201: Simulation resources are not available for the Nav2 Demo

eholum, 2024-10-22:
### Version or commit hash

main

### RMW Implementation

FastRTPS

### Steps to reproduce issue

Run the nav2 demo, the robot model does not appear in the rviz widget because the meshes are not available in the container. This I believe is an old issue unrelated to the recent change, just noticed it while testing.

### Expected behavior

Would like to see the rover for visualization purposes.

### Actual behavior

We only get tf and a geometry error saying the mesh files cannot be found

franklinselva, 2024-10-26:
I didn't realise there is a `nav2` demo. I am expecting the same case for `moveit2` too.

Do you think these demos can be adapter after space-ros/demos#31

eholum, 2024-11-03:
Not gonna worry about this now, since we'll have time to rework that demo before the next release.

closed 2024-11-03 by eholum