space-ros/stacks #208: nav2_dem Time Out on navigation.launch

DionisisGiannar, 2024-12-18:
### Version or commit hash

nav2_demo-timeout-tf_error-odom-base-link

### RMW Implementation

FastRTPS

### Steps to reproduce issue

**Build** with ./build.sh the **navigation2** image: osrf/space_nav2:latest
**Build** with ./build.sh the **nav2_demo** image: osrf/space_nav2_demo:latest
**Run** with ./run.sh the **nav2_demo** image

Inside the container created from nav2_demo 
run the command:
`ros2 launch space_nav2_bringup navigation_launch.py use_sim_time:=True params_file:=nav2_params.yaml`

### Expected behavior

Not expected error and not timed out.

### Actual behavior

Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist

The above error is repeating for ever.

### Backtrace or Console output

[INFO] [launch]: All log files can be found below /home/spaceros-user/.ros/log/2024-12-18-23-01-15-959372-Nebula-700
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [controller_server-1]: process started with pid [701]
[INFO] [smoother_server-2]: process started with pid [703]
[INFO] [planner_server-3]: process started with pid [705]
[INFO] [behavior_server-4]: process started with pid [707]
[INFO] [bt_navigator-5]: process started with pid [709]
[INFO] [waypoint_follower-6]: process started with pid [711]
[INFO] [velocity_smoother-7]: process started with pid [713]
[INFO] [lifecycle_manager-8]: process started with pid [715]
[lifecycle_manager-8] [INFO] [1734562876.111957665] [lifecycle_manager_navigation]: Creating
[lifecycle_manager-8] [INFO] [1734562876.112349755] [lifecycle_manager_navigation]: Creating and initializing lifecycle service clients
[waypoint_follower-6] [INFO] [1734562876.112338053] [waypoint_follower]: 
[waypoint_follower-6] 	waypoint_follower lifecycle node launched. 
[waypoint_follower-6] 	Waiting on external lifecycle transitions to activate
[waypoint_follower-6] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[velocity_smoother-7] [INFO] [1734562876.113340740] [velocity_smoother]: 
[velocity_smoother-7] 	velocity_smoother lifecycle node launched. 
[velocity_smoother-7] 	Waiting on external lifecycle transitions to activate
[velocity_smoother-7] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[smoother_server-2] [INFO] [1734562876.115221732] [smoother_server]: 
[smoother_server-2] 	smoother_server lifecycle node launched. 
[smoother_server-2] 	Waiting on external lifecycle transitions to activate
[smoother_server-2] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[controller_server-1] [INFO] [1734562876.115630493] [controller_server]: 
[controller_server-1] 	controller_server lifecycle node launched. 
[controller_server-1] 	Waiting on external lifecycle transitions to activate
[controller_server-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-3] [INFO] [1734562876.115749795] [planner_server]: 
[planner_server-3] 	planner_server lifecycle node launched. 
[planner_server-3] 	Waiting on external lifecycle transitions to activate
[planner_server-3] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[waypoint_follower-6] [INFO] [1734562876.116191828] [waypoint_follower]: Creating
[bt_navigator-5] [INFO] [1734562876.118586696] [bt_navigator]: 
[bt_navigator-5] 	bt_navigator lifecycle node launched. 
[bt_navigator-5] 	Waiting on external lifecycle transitions to activate
[bt_navigator-5] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[bt_navigator-5] [INFO] [1734562876.118633303] [bt_navigator]: Creating
[behavior_server-4] [INFO] [1734562876.119945055] [behavior_server]: 
[behavior_server-4] 	behavior_server lifecycle node launched. 
[... 66 lines cut ...]
[lifecycle_manager-8] [INFO] [1734562876.253092143] [lifecycle_manager_navigation]: Configuring behavior_server
[behavior_server-4] [INFO] [1734562876.253177011] [behavior_server]: Configuring
[behavior_server-4] [INFO] [1734562876.254223169] [behavior_server]: Creating behavior plugin spin of type nav2_behaviors/Spin
[behavior_server-4] [INFO] [1734562876.255599000] [behavior_server]: Configuring spin
[behavior_server-4] [INFO] [1734562876.256358013] [behavior_server]: Creating behavior plugin backup of type nav2_behaviors/BackUp
[behavior_server-4] [INFO] [1734562876.257641884] [behavior_server]: Configuring backup
[behavior_server-4] [INFO] [1734562876.258196165] [behavior_server]: Creating behavior plugin drive_on_heading of type nav2_behaviors/DriveOnHeading
[behavior_server-4] [INFO] [1734562876.259511734] [behavior_server]: Configuring drive_on_heading
[behavior_server-4] [INFO] [1734562876.260119005] [behavior_server]: Creating behavior plugin assisted_teleop of type nav2_behaviors/AssistedTeleop
[behavior_server-4] [INFO] [1734562876.262419909] [behavior_server]: Configuring assisted_teleop
[behavior_server-4] [INFO] [1734562876.263199590] [behavior_server]: Creating behavior plugin wait of type nav2_behaviors/Wait
[behavior_server-4] [INFO] [1734562876.264413340] [behavior_server]: Configuring wait
[lifecycle_manager-8] [INFO] [1734562876.265077176] [lifecycle_manager_navigation]: Configuring bt_navigator
[bt_navigator-5] [INFO] [1734562876.265147927] [bt_navigator]: Configuring
[lifecycle_manager-8] [INFO] [1734562876.294061616] [lifecycle_manager_navigation]: Configuring waypoint_follower
[waypoint_follower-6] [INFO] [1734562876.294140202] [waypoint_follower]: Configuring
[waypoint_follower-6] [INFO] [1734562876.296050148] [waypoint_follower]: Created waypoint_task_executor : wait_at_waypoint of type nav2_waypoint_follower::WaitAtWaypoint
[lifecycle_manager-8] [INFO] [1734562876.296274545] [lifecycle_manager_navigation]: Configuring velocity_smoother
[velocity_smoother-7] [INFO] [1734562876.296382546] [velocity_smoother]: Configuring velocity smoother
[lifecycle_manager-8] [INFO] [1734562876.297038838] [lifecycle_manager_navigation]: Activating controller_server
[controller_server-1] [INFO] [1734562876.297097497] [controller_server]: Activating
[controller_server-1] [INFO] [1734562876.297114659] [local_costmap.local_costmap]: Activating
[controller_server-1] [INFO] [1734562876.297120831] [local_costmap.local_costmap]: Checking transform
[controller_server-1] [INFO] [1734562876.297133014] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[controller_server-1] [INFO] [1734562876.797197712] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist

DionisisGiannar, 2024-12-18:
Can someone help me with this? 

Furthermore, I did not find any gazebo in install folder or in /usr/share/. Is this suppose to open gazebo too or not?

I also tried to continue the steps from the [README.md](https://github.com/space-ros/docker/blob/main/nav2_demo/README.md) file, and I could not make navigation planning, I made only pose estimation. Maybe this is because of the above Isuue, I am not sure.

mkhansenbot, 2025-01-15:
This looks like it's waiting for the transform from localization. Did you run localization in another terminal like the [instructions in the README](https://github.com/space-ros/docker/blob/main/nav2_demo/README.md#terminal-3---launch-localization-with-map)?

ivanperez-keera, 2025-04-27:
@DionisisGiannar  following up on this.

DionisisGiannar, 2025-04-28:
> This looks like it's waiting for the transform from localization. Did you run localization in another terminal like the [instructions in the README](https://github.com/space-ros/docker/blob/main/nav2_demo/README.md#terminal-3---launch-localization-with-map)?

Yes, I followed instructions step by step.